eez-open/psu-hw #9: Fan generates EMI

prasimix, 2016-10-20:
When cooling fan is active, and additional noise could be measured on the channel's output. We need here better filtering or maybe switching from PWM to linear speed control could improve situation.

PurnenduK90, 2017-01-30:
Can you point out which board is having the fan controller and where is it mounted. As I work on KiCAD/Allegro hence can't open your design files. If you can point out PDF or image may be I can help.

prasimix, 2017-01-30:
It's located on AUX PS module, you can find it on [Sheet 7of11](https://raw.githubusercontent.com/eez-open/psu-hw/master/Consolidated/EEZ%20PSU%20consolidated%20r5B11%20Sheet%207of11.png). Q22 is driven with MCU PWM signal `AUX_FAN_CTRL`.

PurnenduK90, 2017-01-30:
Sorry, I meant PCB implementation, is it properly shielded from the main board. As I don't have the complete picture hence I can only guess. Please look into following points:

1. What is the frequency of noise in main power {Is this same or some multiplier of fan controller switching frequency.}, If yes, you can do following things;

- Provide an isolated ground (1uh inductor)
-Provide an isolated power with sufficient input capacitors
-Make sure switching lines are separated by output traces by ground plane of 3*trace width and followed by ground vias separated maximum by 1/20th of the switching frequency wavelength.
-Heating also cause the rise in noise so please make sure MOSFET is not heating up.
-Physical position of fan can also be the reason of noise due to rotating field (minor)
-For simplicity, a linear controller can also be used.

if the noise frequency is different than fan switching or its harmonics, it means some noise is finding return path through fan circuit. In this case, the above-mentioned steps will still help.

prasimix, 2017-01-30:
Thanks for that. The physical location of both AUX PS module and fan can be see on [picture1](http://www.envox.hr/eez/images/galleries/psu-h24005/02_PSU_H24005_topview_open.JPG) and [picture2](http://www.envox.hr/eez/images/galleries/psu-h24005/07_fan_orientation.JPG) (both of them are mounted directly on the enclosure's rear panel). PCB layout is shown below (fan section is in down right corner as marked with green rectangle):

[image]

[image]

[image]

Fan PWM signal is 1 kHz, while main switcher is on 320 kHz.
I'm thinking to replace existing AC/DC 12V PCB converter (TR2) that is used for powering Arduino shield (using step down IC16, see [Sheet 6of11](https://raw.githubusercontent.com/eez-open/psu-hw/master/Consolidated/EEZ%20PSU%20consolidated%20r5B11%20Sheet%206of11.png)) with one that has two isolated outputs: +5V and +12V. In that case fan could be powered separately and PWM signal can be isolated with adding an opto-coupler. Maybe that is a right approach even if linear controller is used.

prasimix, 2017-01-31:
This is a new proposal for fan driver with isolated power supply. It's based on Vigortronix VTX-214-005-0512 dual isolated outputs power supply (+5V, +12V). Fan control and speed sense is accomplished using two opto couplers.

[image]

PurnenduK90, 2017-02-01:
I think OK5 is conducting more or equal to the maximum specified load current. I will suggest R128 to be replaced by 2.2K. 
I didn't find the spice model for SKL16, although, with another Schottky diode, I found circuit was not regulating properly with fan model I choose. If you can give some specification of fan model I can comment further. Also, fan-sense circuit is not very clear to me. Just curious, what is output of fan sense (a constant voltage based on fan speed/ PWM based on fan speed).

prasimix, 2017-02-01:
With existing R128 max. Ic on OK5 is limited below 6.5 mA. That is well below max. rating from HCPL-151 datasheet (30 mA @ If=5 mA). I also found that fan sense output max. Ic=4 mA therefore I added Q13 as on picture below. Fan is [SUNON MB60251V1-000U-G99](http://www.tme.eu/en/details/mb60251v1-g99/dc12v-fans/sunon/mb60251v1-000u-g99/).

[image]

PurnenduK90, 2017-02-01:
yes, you are correct I misread the exponent value. This circuit seems fine although I would like to increase the value of c97 to 10m to filter out the ripple. As fan speed control is not required instant task so I will not mind the soft change in voltage level.

prasimix, 2017-02-01:
Huh, 10m for C97 looks quite challenging on existing PCB :). I'll see what is the biggest possible package that can be included.

PurnenduK90, 2017-02-01:
what package is used in current board

prasimix, 2017-02-01:
Ø6.3 x 7.7mm

prasimix, 2017-02-13:
VTX-214-005-0512 is tested under full load on both rails and it works fine. I'll continue when new PCB arrive.

prasimix, 2017-03-14:
It seems that isolated fan power did the job.

closed 2017-03-14 by prasimix