eez-open/psu-hw #14: Isolate TPS3705 reset signal

prasimix, 2016-12-11:
Arduino IDE cannot reset the board when firmware is uploaded and reset has to be applied after uploading attempt. The reason for that is that TPS3705 is holding RESET line high (inactive). That makes a whole uploading procedure longer (and frustrating). Add one diode to "isolate" TPS3705 reset pin from Ardino Due RESET line.

prasimix, 2016-12-11:
Fixed

closed 2016-12-11 by prasimix

algaremi, 2016-12-13:
Do you have a schematic for this mod? I would like to add it to my Arduino shield. Thanks
SB

prasimix, 2016-12-13:
This one is not trivial, it require that IC32 is removed that one trace beneath it can be cut. Also one 0.8 mm hole has to be drilled on the right position. Use any THT signal diode e.g. 1N4148:

[image]

[image]

[image]

bgat, 2016-12-13:
Why not just lift pin 7 on IC32?

prasimix, 2016-12-13:
Good point, thanks! :+1:

bgat, 2016-12-13:
Wait, nevermind... 😂

bgat, 2016-12-13:
Wait, I take that back...once you lift pin 7, and just looking at the pics (and not schematics, which I don't have in front of me), it appears you might have a couple of vias you could solder the diode to. So maybe the pass-thru hole isn't necessary.

I belabor the point only because drilling is irreversible, and if a noob (like me when I'm under caffeinated) makes that kind of mistake, suddenly they're into a much bigger problem...

prasimix, 2016-12-13:
I understand that. If you really want to be on the safe side, forgot this hack. Only obstacle is that you need to push reset manually while firmware is uploading.

bgat, 2016-12-13:
Nope, I want automatic AND safety. Call me lazy, or call me preparing for someone like me who won't read directions. 😂

prasimix, 2016-12-13:
:) that's also fine. That's why this issue is opened and fixed in the recent revision.