eez-open/psu-hw #22: Suggestions

PurnenduK90, 2017-01-30:
After changing to subscription based licensing I think this project should move to KiCAD. It has a lot of potentials. I will also suggest some fixed outputs to easily work with digital design. e.g. 5V, 3.3V, 2.5V, 1.25V. As an electronics engineer, I never find one power supply enough for testing purpose, due to multiple voltage levels required in modern designs.

prasimix, 2017-01-30:
Please open one issue per suggestion in the future.

prasimix, 2017-01-30:
For multiple outputs we should in the future replace existing AC/DC converters (2 x 48V, 1 x 12V) with one **open source** LLC converter ([example](https://github.com/yu3ma/FSFR2100_LLC_SMPS)) with multiple isolated outputs. 
Another possibility is to go with "rackable" design with backplane that can grow to at least 8 different test&measurements modules with central TFT console.

prasimix, 2020-01-20:
For multiple outputs check [EEZ BB3](https://github.com/eez-open/modular-psu) project.

closed 2020-01-20 by prasimix