eez-open/psu-hw #35: BOM Errors?

jwilhelm1, 2017-03-24:
R18 on the power boards isn't listed in the BOM
The listed inductor for L1 doesn't fit the board footprint. RN114-4-02-1MS does fit.

prasimix, 2017-03-24:
R18 is missing, I'm going to add it, but I'm not so sure about L1. It's changed from 1.5mH to 0.7mH that is much smaller but both can be used (it's now smaller due to #25). Please let me know what Power board revision you are referring at?

jwilhelm1, 2017-03-24:
The power board I am assembling is 02/2017 r5B12.

prasimix, 2017-03-24:
OK, but r5B12 is still open for revisions and you can see that few builds exists on the GitHub (started from Feb 13th build d9b361a). That will be like that until it become an official new release that will be listed [here](https://github.com/eez-open/psu-hw/releases).  
It's quite possible that you used some of earlier builds and have the following situation:

[image]

jwilhelm1, 2017-03-24:
Thanks. I'm new to GitHub, so I'm still learning how to navigate it.

closed 2017-03-24 by jwilhelm1

prasimix, 2017-03-24:
No problem, and many thanks for your patience and contribution.