eez-open/psu-hw #47: Unit won't power on

madscientist159, 2017-08-13:
Received kit today, set both internal PSUs to 115V (local US voltage), assembled it, eagerly plugged it in and ... nothing.  No lights, no fan, nothing.

Double checked all wiring already; will start debugging shortly.  Has anyone else experience this?  Note that on my unit the rear power switch did not easily go into the "On" position; I don't know if I may have received a unit with a defective rear power switch?

madscientist159, 2017-08-13:
And in case anyone asks, yes, the fuse tray is installed with both fuses :wink:

madscientist159, 2017-08-13:
And the verdict is...defective mains switch :nauseated_face:   One pole doesn't close in the "On" position.

Can I just get a replacement switch sent out?

madscientist159, 2017-08-13:
Starting to wonder if you guys shipped a defective rear panel.  With mains switch bypassed unit enters boot loop; seems like main PSU can't supply enough power to keep front panel / control electronics engaged.  Unit tries to start up, runs out of stored energy in PSU caps, then resets and tries to start up again.

Not exactly a happy camper here. :-1:

madscientist159, 2017-08-13:
Uplugging fan fixed startup issues but...no fan means unit is unable to be installed on bench.  This rear panel assembly seems to be quite messed up; can you overnight a new one?

prasimix, 2017-08-13:
Hi, as mentioned in the _[kit assembly instructions](http://www.envox.hr/eez/bench-power-supply/group-buy-kit-assembly.html)_ PCB modules are tested independently of accompanying parts therefore AUX power module board is tested without all parts mounted on the rear panel. But, all parts are genuine and installed as received from the distributor (e.g. TME in case of main switch). It's possible that main switch is ruined while wires are inserted in it. But, that cannot explain why AUX power module doesn't work. Please first check without any load if you can get +5 V and +12 V on its outputs.

madscientist159, 2017-08-13:
Interesting that you mention that.  Just checked under load and the "12V" output reads 17V (!) while the 5V output is a stable 4.95V.

FWIW I am an electrical engineer with many, many years experience with systems that are more complex than this one.  The switch problem I can sort of understand, what I don't understand is how this unit passed testing with a faulty standby regulator module.

madscientist159, 2017-08-13:
@prasimix Any idea of when I can get a replacement for the failed module?  We have rework facilities here so if it will be faster to just send out a switch module and a replacement VTX-214-005-0512 I'm OK with that.

prasimix, 2017-08-13:
If I understand you correctly TR1 +12 V rail is defective. But, even if that is a case, with proper +5 V rail your unit should start and power up self-test will report fan error. Please let me know what you get in you case.

madscientist159, 2017-08-13:
@prasimix That is exactly what happens.  I don't want to run the unit w/o a fan since the two power supply boards contain linear regulators and I get concerned about overall heat dissipation.

prasimix, 2017-08-13:
Ok, overheating cannot happen since firmware when detecting fan error limit max. output current to 2 A (defined by `ERR_MAX_CURRENT` constant).

madscientist159, 2017-08-13:
@prasimix Good to know.  Still, I purchased a kit that was supposed to be fully functional and I need more than 2A on the outputs.  How soon can I get replacements for the two defective components?

prasimix, 2017-08-13:
I cannot answer in this moment how soon you can get replacement. Please note that we are not factory, and that this campaign was actually a group buy. Therefore no extra boards nor service parts are available. I have to order it, and that will take time. Also I cannot initiate a new order as someone report a problem, but I have to combine parts from more incidents (if that will happen) to save shipping costs.

madscientist159, 2017-08-13:
@prasimix All right then, how about a partial refund and I'll just source the parts?

prasimix, 2017-08-13:
That's also possible, I have to discuss that with Crowd Supply first. Give me a day or two.

madscientist159, 2017-08-13:
@prasimix OK, thanks! :smile:

prasimix, 2017-08-14:
Please send me your PayPal email using contact [form](http://www.envox.hr/eez/contact-us.html).

madscientist159, 2017-08-14:
@prasimix Email sent.

closed 2017-08-14 by prasimix

madscientist159, 2017-08-30:
Need to add one more defect to the list...this power board was a mess.  In addition to everything else (causing the module failure?) the fan power leads were swapped inside the fan connector!  Replacing the module and manually flipping the red / black wires appears to have restored the rear panel to operation, a couple of weeks late and with a $20 extra investment in replacement parts.

Overall not happy here...won't order again or recommend until the rear panel gets better QA.