eez-open/psu-hw #50: EEZ H24005 Early Bird Kit Parts Shortage & Assy documentation errors

stolens, 2017-08-18:
EEZ H24005: Open Source Bench Power Supply Kit Early Bird Complete kit Parts Shortage.

There is a fulfillment shortage and documentation errors in the assembly manual for this kit.

Reference:
Complete kit EEZ H24005: Open Source Bench Power Supply
http://www.envox.hr/eez/bench-power-supply/group-buy-kit-assembly.html
A work in progress (Draft 0.7)

Parts group Bag#2, part B1. Parts list indicates only 2 pieces of M3x5 mm Black, Countersunk head screws when 4 are required causing a parts shortage in the kit. These are used in mounting the AC/DC Modules to the bottom enclosure. These are illustrated and described around Figures 16 and 17 in the assembly instructions.

Recommended corrective actions:
1. Provide 2 more of part B1 in the kit.
2. Correct the assembly instructions to indicate 4 pcs instead of 2 pcs in bag#2, the mounting hardware for the power modules. The assembly instructions are correct, the provided hardware and the parts list are deficient.

Crowd Supply recommended I post this as an issue.

prasimix, 2017-08-18:
Due to logistic issues we cannot organize sending of two additional screws. Sorry for that.  
But as you can see the main mean of fixing and securing AC/DC modules are mounting frame (B3) _not_ 40 mm spacers (B2). Therefore you can left lower side of spacers unfixed, or if you really want to do that you have a great freedom to chose some other M3 screws (length could be up to 30 mm!).

Kit assembly instructions document with this note is now also available [here](http://www.envox.hr/eez/bench-power-supply/group-buy-kit-assembly.html).

szwesleyzhao, 2017-08-19:
Cool, do you need turnkey pcb assembly service in china ? We could one stop stop service from prototype to mass production. Here is email; szhwpcba@gmail.com

davidcarne, 2017-08-26:
Regarding securing the spacers - I suggest against leaving one of the spacers secured with only the top screw. The spacer near the regulator module is not captive as the meanwell module doesn't have a full hole; only a notch. If the module is shaken or the screw/standoff connection isn't quite tight enough, it might shake loose leaving the standoff and screw loose in the enclosure. (This happened during my final assembly).

closed 2018-01-17 by prasimix