eez-open/psu-hw #52: Difficult or unable to attach power modules to case...

mloftis, 2017-08-26:
On my unit the front panel PCB is bowed out slightly making the screw holes for the heatsinks not line up.  Looks to be about a ~1mm difference between the left (screen) side and right (output jacks).  The mounting plate for the screen also barely matches up with the screw holes on the bottom/through the case.

mloftis, 2017-08-26:
This appears to be due to too much overall length created by the jacks themselves.  It's JUST barely enough that the holes miss on the bottom of the case.

prasimix, 2017-08-26:
Please provide picture and mark questionable section.

prasimix, 2017-08-26:
Please also consider reply [#732](http://www.eevblog.com/forum/projects/diy-programmable-dual-channel-bench-psu-0-50v3a/msg1288779/#msg1288779) on the EEVblog forum.

mloftis, 2017-08-27:
Lighting in my office is terrible so it's really hard to see the bottom of case picture.  The screw holes just barely peek through the case holes no matter how you wiggle/maneuver the PSU modules if the front panel is secured.  But the top picture of the front panel alignment issue seems pretty obvious even in the pictures.  That's what's causing the misalignment of the PSU board heat sinks.  I was able to get it to assemble by very loosely attaching the front panel and then attaching the PSUs - I didn't see your post about the EEVblog post until after I'd attempted that way but that's exactly the workaround they came up with too.

[image]
[image]

prasimix, 2017-08-27:
OK, did you still have a problem or not? Does very loosely attaching the front panel before Power boards are mounted on bottom plate resolve this issue?

mloftis, 2017-08-28:
Yes, loosely attaching the front panel in order to gain the ~1mm movement to get the holes aligned worked.

closed 2017-08-28 by mloftis