eez-open/psu-hw #53: issue with USB port

jlegen, 2017-08-26:
Hi,
i successfully assembled my psu device, values (without load and calibaration) are basically bang-on.
The only thing which i don't get working is the USB connector. Native port is only recognized as "device not started, Code 10" in device manager.
I tried several cables, and also checked that the internal cable is ok, and it's wired correctly.
I managed to update to firmware 1.01 by plugging a USB cable into DUE's native port directly, which means port and windows (driver) are ok.
I first thought it could only be a simple mechanical issue, but then i realized that it's got an isolator before the USB port... 8-}
Any ideas on that?

Update: is just tried to re-wire to DUE's programming port - and this works! I've been able to re-flash the firmware configured to prog-port as well, so at least it works "some how". 
Though the constant resetting of the whole thing is really annoying (played a bit with SCPI)...

Jan

prasimix, 2017-08-26:
Ok, so what is your problem here? Does USB isolator and provided USB patch cable (5-pin 0.1" to micro-USB) works with _programming_ but not with _native_ port on Arduino Due?

jlegen, 2017-08-26:
Yes. 
It just took me a while until i came to the idea to try the prog port, instead of native.

jlegen, 2017-08-26:
little FUP on this - according to USB-Tool, the native port's ID is read correctly, error happens afterwards; perhaps some out-of-spec passive at the Adum, resulting in some form of edge/corner case? 

Device Description       : USB-Verbundgerät
Device Path              : \\.\USB#VID_2341&PID_003E#5&1c6e4acc&1&1#{a5dcbf10-6530-11d2-901f-00c04fb951ed}
Device ID                : USB\VID_2341&PID_003E\5&1C6E4ACC&1&1
Driver KeyName           : {36fc9e60-c465-11cf-8056-444553540000}\0015 (GUID_DEVCLASS_USB)
Driver                   : C:\WINDOWS\System32\drivers\usbccgp.sys (Version: 10.0.14393.0  Date: 2006-06-21)
Driver Inf               : C:\WINDOWS\INF\usb.inf
Legacy BusType           : PNPBus
Class                    : USB
Service                  : usbccgp
Enumerator               : USB
Location Info            : Port_#0001.Hub_#0002
Location IDs             : PCIROOT(0)#PCI(1A01)#USBROOT(0)#USB(1), ACPI(_SB_)#ACPI(PCI0)#ACPI(USB4)#ACPI(URTH)#USB(1)
Container ID             : {e010817c-7873-11e7-98f3-001e3790fa1d}
Manufacturer Info        : (Standard-USB-Hostcontroller)
Capabilities             : Removable
Address                  : 1
Problem Code             : 10 (CM_PROB_FAILED_START)
Power State              : D3 (supported: D0, D2, D3, wake from D0, wake from D2)

      ------------------ Device Descriptor ------------------
bLength                  : 0x12 (18 bytes)
bDescriptorType          : 0x01 (Device Descriptor)
bcdUSB                   : 0x200 (USB Version 2.00)
bDeviceClass             : 0x00 (defined by the interface descriptors)
bDeviceSubClass          : 0x00
bDeviceProtocol          : 0x00
bMaxPacketSize0          : 0x40 (64 bytes)
idVendor                 : 0x2341
idProduct                : 0x003E
bcdDevice                : 0x100
iManufacturer            : 0x01
 *!*  string descriptors are not available because device is in low power state
iProduct                 : 0x02
 *!*  string descriptors are not available because device is in low power state
iSerialNumber            : 0x03
 *!*  string descriptors are not available because device is in low power state
bNumConfigurations       : 0x01

ConfigDescriptor Error   : ERROR_GEN_FAILURE

prasimix, 2017-08-26:
Hm, not sure about "edge/corner" case, do you have by chance some other Arduino Due board to test it?

jlegen, 2017-08-27:
Tried with another Arduino Due - exactly the same behaviour - only prog port working with isolator.
So imo it could be something with
- internal micro USB header cable (imo unlikely, since it works with prog port (?))
- USB port at case's back (imo unlikely, too)
- Adum isolator

prasimix, 2017-08-27:
You're right, and I believe the first two case should be easy to check. The latest one will require new IC or some scope measuring and probing. Maybe it's also possible that some e.g. "dirt" or other irregularity can be found on the PCB with simple visual check.

jlegen, 2017-08-29:
Dismounted the pcb at the back, it looked absolutely nice and clean. Resoldered Adum and surrounding passives anyway, and cleaned all with isopropanyl afterwards. No change... :-/
Probably unrelated, but i noticed while checking continuity for USB connection:
- while the USB housing at the back is grounded to earth, the metal front metal bezel for the display is not (front metal plate has contact, though)
- the "shield" pin (5) of the USB connection cable to the back pcb seems not to be connected to anything (seems ok according to schematics?)

prasimix, 2017-08-29:
Correct, neither of them are connected to the earth.

jlegen, 2017-08-30:
i soldered a new micro USB header cable, just to be sure; same result/no native port; running out of ideas - apart from throwing a scope onto it... 8-}

prasimix, 2017-08-30:
If it works flawlessly with _programming_ and not _native_ port then only proper port selection under **Tools... Board** menu is missing. No other ideas so far.

jlegen, 2017-09-01:
I did some further "research" - i have an external USB isolator module, which happens to use the exact same Adum3160.
And guess what, this thing behaves exactly the same like the EEZ-PSU, with PSU's Due and also with another standalone Due board. 

It also turned out that "software" doesn't matter - the same Win10 system which does not work with Adum+Due-native does also not work with a current Linux on the same hardware.

What is however interesting - while the "Due Prog port" through Adum is detected as "USB Fullspeed" (= USB 1.1), the native port is "USB Highspeed" (= USB 2.0). This at least could explain why the Adum behaves differently for prog and native port...

I also found some sort of solution - i dug out a very old laptop which i kept for, well, things like that :), and this laptop is able to properly work with Due's native port with the Adum in between.

Knowing this, i then attached the Due-native to an USB-2 port sitting directly on the mainboard of my Win10 system (which did fail before, using the front panel ports)  - and this setup also worked!

So the outcome probably is - the Adum simply seems to be a bit picky regarding the USB standard. Avoiding USB ports broken out with headers or hubs seem to help at least sometimes. 
My regular laptop however - the system which i use in the lab - still does not work, and "swapping" USB ports is obviously not possible there.

But i think i will now leave the PSU at native port, doing SCPI-tests and fw updates with the old laptop (XP, uh.. :o) mentioned above...

prasimix, 2017-09-01:
Thx, for additional info. Regarding SCPI testing please note that you can do it over Ethernet too.

closed 2020-01-20 by prasimix