eez-open/psu-hw #54: self-test not passed

ulaker, 2017-08-27:
access to reset switch obstructed by display board.
Front pannel bottom screws misregistration difficulties to allign threaded part with the case.
Unkeyed and unmarked pin connectors no easy way to determine right orientation. 
First power up resulting in message self-test not passed for sd-card, rtc, both channel ioexp and adc
Troubleshooting guide not mentioning  path to follow for these cases.

prasimix, 2017-08-27:
This is four, not one issue. Please don't do that next time since that makes it difficult to follow up and other people will have difficulties to follow and could add duplicated issues.

> access to reset switch obstructed by display board.

I'm aware of that, but currently there is no remedy for it. If you need reset in the future you still have enough room to access it with sort of paperclip.

> Front pannel bottom screws misregistration difficulties to allign threaded part with the case.

Duplicate, see #52 

> Unkeyed and unmarked pin connectors no easy way to determine right orientation.

Please elaborate. Also consult Fig. 13 and Fig. 24 ([Draft 0.9](http://www.envox.hr/eez/bench-power-supply/group-buy-kit-assembly.html)).

> First power up resulting in message self-test not passed for sd-card, rtc, both channel ioexp and adc
> Troubleshooting guide not mentioning path to follow for these cases.

SD card is possibly not plugged in correctly, check it once again, if still have report error, try to test it on PC/notebook that is equipped with SD card reader.
For Power boards (channels) error, check once again if they are plugged correctly into Arduino Board (sometimes is possible that connectors are shifted up for one position). If you still have a problem, start with checking channel input power (connector X1) where you should have +48 Vdc, and then bias voltages on the output of IC4 (position FB1, FB2, and FB3).

ulaker, 2017-08-27:
Sorry, new here. f
The cable connectors seem sorted now. part  my fault confusion from so many places where I thought it is documented. 

From my perspective the mechanical issues are cosmetic and can be closed as far as I am concerned since I do not expect changes to apply to my kit.

tried to narrow it down by disconnecting sd-card and power boards, still get the rtc self-test not passed. The communication on the spi bus is seemingly not working. Will take some time to locate components that could be detached to isolate the problem.
  
SD card tested separately and seems to be formatted and working on a different environment. 

Power +48V nominal on both boards available.

prasimix, 2017-08-27:
Ok, try to unplug Power boards and power up just Arduino shield and see if e.g. RTC selftest error is still there.

ulaker, 2017-08-27:
Done that. Also disconnected sd card.
Still no response from rtc 
Home menu reachable tried to set time/date also error.

prasimix, 2017-08-27:
If SPI communication is an issue, did you also receive EEPROM seltest error?

ulaker, 2017-08-27:
Initially for several iterations of power up sequences the eeprom was there 
Since early this morning no longer in the list of the self-test not passed.

prasimix, 2017-08-27:
Hm, then it's possible that something related to RTC. You can try to upload test sketch that set date and time and display on the serial monitor new date/time every one second.

``` arduino
#include "SPI.h" // necessary library
#include <eez_psu_rev.h>
#define EEZ_PSU_SELECTED_REVISION EEZ_PSU_REVISION_R5B12 // change to EEZ_PSU_REVISION_R1B9 for older revision
#include <eez_psu.h>

#define RD_CONTROL_1 B10010000
#define WR_CONTROL_1 B00010000
#define RD_CONTROL_2 B10010001
#define WR_CONTROL_2 B00010001

#define RD_SECONDS B10010010
#define WR_SECONDS B00010010
#define RD_MINUTES B10010011
#define WR_MINUTES B00010011
#define RD_HOURS B10010100
#define WR_HOURS B00010100

#define RD_DAYS B10010101
#define WR_DAYS B00010101
#define RD_WEEKDAYS B10010110
#define WR_WEEKDAYS B00010110
#define RD_MONTHS B10010111
#define WR_MONTHS B00010111
#define RD_YEARS B10011000
#define WR_YEARS B00011000

#define RD_CLKOUT B10011101
#define WR_CLKOUT B00011101

unsigned int seconds;
unsigned int minutes;
unsigned int hours;
unsigned int day;
unsigned int weekdays;
unsigned int month;
unsigned int year;

void setup()
{
  eez_psu_init();

  // Initialize serial communication
  Serial.begin(9600);
  // wait until the serial stream is not open
  while (!Serial);
  Serial.println("Serial com ready");

  SPI.begin(); // wake up the SPI bus
  SPI.beginTransaction(PCA21125_SPI);
  digitalWrite(RTC_SELECT, HIGH); // Select PCA21125
  SPI.transfer(WR_CONTROL_1);  // Write mode, a pointer to the address 00h
  SPI.transfer(B00001000);
  SPI.transfer(B00000000);

  // Set hours, minutes and seconds to 16:10:00
  SPI.transfer(B10000000); // Seconds=00; RF - clock integrity is not guaranteed
  SPI.transfer(0x00); // Minutes
  SPI.transfer(0x12); // Hours, 24-hour format

  // Set day, day of week, month and year to 17.07.2015
  SPI.transfer(0x23); // Days
  SPI.transfer(0x01); // 5, is Firday
  SPI.transfer(0x05); // Months
  SPI.transfer(0x16); // Years
  digitalWrite(RTC_SELECT, LOW); // Deselect PCA21125

  delay(2);

  digitalWrite(RTC_SELECT, HIGH); // Select PCA21125
  // Read day, month and year
  SPI.transfer(RD_DAYS);
  day = SPI.transfer(0x00);
  weekdays = SPI.transfer(0x00);
  month = SPI.transfer(0x00);
  year = SPI.transfer(0x00);
  digitalWrite(RTC_SELECT, LOW); // Deselect PCA21125
  SPI.endTransaction();

  // Display received date
  // HEX format is used since results are BCD encoded
  Serial.print("20");
  Serial.print(year, HEX);
  Serial.print("-");
  Serial.print(month, HEX);
  Serial.print("-");
  Serial.println(day, HEX);
  Serial.print("Day of week:");
  Serial.println(weekdays, HEX);

}

void loop()
{
  SPI.beginTransaction(PCA21125_SPI);
  digitalWrite(RTC_SELECT, HIGH); // Select PCA21125
  SPI.transfer(RD_SECONDS); // Read mode, a pointer to the address 02h
  seconds = SPI.transfer(0x00);
  minutes = SPI.transfer(0x00);
  hours = SPI.transfer(0x00);
  digitalWrite(RTC_SELECT, LOW); // Deselect PCA21125
  SPI.endTransaction();

  // Display received time
  // HEX format is used since results are BCD encoded
  Serial.print(hours, HEX);
  Serial.print(":");
  Serial.print(minutes, HEX);
  Serial.print(":");
  Serial.println(seconds, HEX);
  delay(1000);
}

ulaker, 2017-08-27:
Sorry not in the position to follow these instructions immediately. First I need to set up and get working the development environment.

jlegen, 2017-08-27:
Regarding SD card - i also had it failing on power-up after i upgraded to FW 1.01. It turned out that i only had the '2nd newest' SD library on my system. Upgrading by IDE's library manager solved that.

prasimix, 2017-08-27:
@ulaker: The easiest way to get all needed headers and files is to upgrade your firmware to 1.01. Follow instructions [here](http://www.envox.hr/eez/bench-power-supply/building-instructions.html#instr_fw) and mind discussion about external libraries [here](https://github.com/eez-open/psu-firmware/issues/159).

@jlegen: Please provide more information what you actually did, and which Arduino IDE you are using, that I can include that into assembly instructions.

jlegen, 2017-08-27:
- I installed Arduino IDE 1.8.4
- grabbed firmware 1.01 and both libs from eez psu github
- grabbed ethernet2 from your other gihub
- compiled 1.01 - no errors
- uploaded firmware (worked using prog port)
- psu selftest: SD card error
Since this did not give an error with the pre-installed FW 1.00, i was pretty sure that the problem was on my side. Looked into the eez psu sketch and found, that this self test failure is generated right with/after creating the SD object. 
Then i had a look into the library manager of the arduino IDE and noticed an upgrade for SD lib from 1.1.0 to 1.1.1 (if i remember correctly) - upgrade, recompile, reflash, error gone!

My guess: IDE 1.8.4 is delivered with SD 1.1.0, which seem to have an issue, so they provided an updated version 1.1.1 right after. 
Usually, the IDE also shows a notice for upgrades, but this may take a while from my experience....

ulaker, 2017-08-27:
Hello prasimix,

found some time and managed to compile the rtc test sketch that you provided. Looks the rtc is responding with settings and is updateing the time in one second intervals. This I tested with the arduino and the  front pannel assembly detached and powered from the usb port of the development pc. 

So this would mean the spi communication to the rtu is ok at least as long as no other peripheral attached. 

Of course i lost the factory delivered  firmware.
 The development environment seems not set up correctly since github version of the firmware is not compiling without error 

Arduino: 1.8.4 (Linux), Board: "Arduino Due (Programming Port)"

/home/erich/Arduino/eez_psu_sketch/eez_psu_sketch.ino:31:23: fatal error: Ethernet2.h: No such file or directory
 #include <Ethernet2.h>
                       ^
compilation terminated.
exit status 1

ulaker, 2017-08-27:
found and installed one Ethernet2 library, probably not the right one since compilation aborts with:
Arduino: 1.8.4 (Linux), Board: "Arduino Due (Programming Port)"

sketch/ethernet.cpp: In function 'void eez::psu::ethernet::init()':
ethernet.cpp:146: error: 'class EthernetClass' has no member named 'setDhcpTimeout'
     Ethernet.setDhcpTimeout(ETHERNET_DHCP_TIMEOUT * 1000UL);
              ^
exit status 1
'class EthernetClass' has no member named 'setDhcpTimeout'

this is way above my level at this time.

electrokean, 2017-08-27:
@ulaker use latest Ethernet2 library from here: https://github.com/eez-open/Ethernet2/releases
See discussion on your error here: https://github.com/eez-open/psu-firmware/issues/159

ulaker, 2017-08-27:
thanks, this did indeed help with compiling and installation of firmware version 1.01
unfortunately power up resulting in message self-test not passed for sd-card, rtc, both channel ioexp and adc did not change. 
In addition playing with the menues enabled ethernet and am no stuck probably with the startup not getting past the dhcp since I do not have a dhcp server to help get past this stage in the power up process. Pressing reset does not help to get out of the loop. Is there another whay to force to continue without ethernet when it is enabled?

ulaker, 2017-08-27:
EUREKA some magic worked while I was typing the above. The ethernet setup must have timed out and now it seems to be in a happily working (uncalibrated) state. 
I am now working up and exploring all the options. 
But first I will redo the mechanical assembly and hope to be able to do everything else with all the screws in place.

Thank you all for prompt support and the patience and understanding you had with me.

ulaker, 2017-08-28:
Well, one night later back to square one? After getting excited yesterday late at night with the success in running up the new toy got back to self-test not passed upon switching on this morning. Looks like something is at odds with the timing when and how long the power up self-test routine decides upon pass/fail? More work required in this area I am afraid. 
[image]

prasimix, 2017-08-28:
It's very highly unlikely that you have a "timing issue". The PSU can continuously works for days and weeks without any "timing issue". It's more likely that you have a problem with some connectivity between Arduino Due and Arduino Shield, especially 6-pin header in the middle where are SPI lines. Inspect it and eventually apply some "contact/degreaser" spray such as [KONTAKT WL](http://www.tme.eu/en/details/wl_200/cleaning-maintaining-products/kontakt-chemie/71009-002/).

ulaker, 2017-08-28:
disassembled front pannel and inspected the arduino,  shield and the display module. SD-card socket is a bit flimsy, at least the card does not seem to be firmly hold in place. But I assume it is good enough for the purpose here. The SPI connector also looks optically and electrically (continuity) ok. re-assembled and now EEPROM is no longer on the list of the failed self-test items (CH2 ADC also visible now) Next I would capture some traffic on the bus but I am not equipped here with the required toys.

prasimix, 2017-08-28:
SPI connector in the middle just carry general SPI signals: CLK, MOSI and MISO. Other important signal is "chip select" and each SPI peripheral has it's own line and they are distributed all around as you can see on [Sheet 9](https://raw.githubusercontent.com/eez-open/psu-hw/master/Consolidated/EEZ%20PSU%20consolidated%20r5B12%20Sheet%209of12.png).

ulaker, 2017-08-28:
Another question is the calibration password. When I enter it correctly  it immediately  ask for new and also repeat the new assword? I do not want to change I actually tried to exercize the routine to calibrate the channel. Is it compulsory chnage password associated with that?

prasimix, 2017-08-28:
You want to say that when you enter Calibration wizard that it ask you to change password? I'm not aware of that "feature" :) Can you provide some short video?

ulaker, 2017-08-28:
Maybe it is misunderstanding on my side. The system settings menu item **channel calibration** might be just the password management only and not the  **calibration wizard**. That one is possibly now behind the big red ERROR part. I could technically capture a video clip, but I need to find some fixture to hold and point  the camera to do that, maybe later, in this case it is probably not useful.

prasimix, 2017-08-28:
Ok, now I understand, no video is needed. 
The system settings menu include both calibration password change and each channel calibration options. If you select "Calibration password" change you'll be able to set new password. Please note that will be the _only_ option available/visible on that page if both channels failed to pass selftest.

ulaker, 2017-08-28:
Thank you for patiently educating me. Would take much longer to find this all out myself by reading it somewhere. I might try to find a trypod or improvize some fixture to be able to forward vidoe of the somewhat un-deterministic  behaviour on repeatedly pressing reset butoon (the one on the upper corner of the arduino board. In a substantial number of iterations of the startup from reset it seems to reach the point where it asks if I want to do channel calibration any touch of the screen than transits to standby and black screen followed by startup with self-test error sometimes with jus the channels and most of the time rtc and sd-card also

prasimix, 2017-08-28:
You have to check all connections. When everything is in place you can reset/restart it as much as you want and it should pass selftest every time without exception.

ulaker, 2017-08-30:
Did not have much time yesterday to dedicate to the hunt for this erratic startup behaviour. Today I dismanteled the front pannel assembly again. This time completely and also took out the display, since upon carefull inspection I did not find any problem to which I could attribute what I observed when powering up the unit. 

To my surprize I found this:

[image]

One of the four corners held by screws did not have a spacer and the pins touch the frame. Hope that this did not inflict permanent damage to any of the concernet components.

prasimix, 2017-08-30:
That should be easy to check. What that plastic bag is doing there?

ulaker, 2017-08-30:
That bag holds the screws that I took out on dissassembling the unit.
It attached itself  to the magnet that I used to make the display stay upright tho take the picture in the cover of the psu.

ulaker, 2017-08-30:
Looks promising on power up after assembling the parts again. Will continue to explore and learn about the features.

Thank you all again for assistance.

ulaker, 2017-08-30:
The observed anomaly is explainable with the shorting of bus pins to the display frame. The fix was putting 5 perspex washers is place of the missing metal spacer. This keeps the bisplay board enough away from the frame. After putting all parts in place and all screws tightened no more problems with startup observed.  This area still looks suboptimal bit I do not expect re-occurrence of the problem.  As far as I am concerned this issue can be closed.

closed 2017-08-30 by prasimix