eez-open/psu-hw #57: X12 X14 Pin confusion

dagema, 2017-09-14:
I am confused/concerned about the schematic references to connectors X12 and X14.  According to the schematic, pin X12-8 is connected to AUX1_USET_IN, but in fact it is (properly labeled on the case) connected to (external trigger) GND, which according to the schematic should be connected to X12-1 (casing label "Remote prog.").  Similarily, X12-4 and X12-5 (Sense+ and Sense-) appear to be reversed when compairing actual wiring with the schematic. Also pin X12-6 does not go to R177 /unless you count the pins from right to left).  The same also apply for the X14 connector.  Summarizing, its seems that "Remote prog" and "Ext. trigger" are reversed in the schematic as well as "Sense+ with "Sense-",  for both X12 and X14. I am referring to the "EEZ PSU consolidated rB12.pdf" file for the schematic references, and the  video demonstrating the "External/remote voltage programming" which infers X12 pin numbering as going from 1 to 8 from counting from left to right.  So now I ask, what is right? The schematic pin assignments and labeling match if you count from right to left, but the that does not seem to match the video, which obviously seems to work as expected and looks as if counting from left to right!  Very confusing.

prasimix, 2017-09-14:
Please note that both X12 and X14 are mounted on the bottom (reverse!) side of the PCB. Therefore pin numbers are mirrored. What is marked on the front panel over push-in 8-pin connectors are correct. The easiest way to check it is by activating remote sensing. If that terminals are swapped reverse polarity detection circuit (Q18, Q20 on Sheet 5/12) will instantly generate error!

dagema, 2017-09-15:
OK, so if I understand you correctly, the labelling on the case reads from left to right, so that pin 1 according to casing label is actually pin 8 according to schematic. If that is true, then I am now clear and satisfied.

closed 2017-09-15 by dagema