eez-open/psu-hw #58: Minor: one connector orientation unclear from assembly instructions

raplin, 2017-09-21:
Wow just built mine in an hour and.. perfect!  Extremely impressed.  Anyway, very minor thing; unless I missed it in the assembly instruction PDF, the orientation of unpolarized header A2 where it connects to  the front panel board is not at all obvious. If you zoom in on one of the photos you can just about see which way up it is, but clarification would be useful.
Brilliant kit - very professional

prasimix, 2017-09-21:
Yes, that detail is shown in lower right corner on[ Fig. 24](http://www.envox.hr/eez/images/arduino_shield_cables2.jpg). Feel free to suggest clarification and I'll add it on the web page and PDF doc.

raplin, 2017-09-21:
Ah yes you're right I didn't see the inset image there. Thanks! I'm really impressed with this PSU - the user interface and functionality is even better than I was hoping for.

closed 2017-09-21 by raplin