eez-open/psu-hw #70: Expected self test failures

jfitter, 2020-01-18:
I am looking for some advice without having to wade through all of the code in minute detail.
I have built the PSU except for the power modules. For testing I am powering the Arduino Shield from the Native USB port. This enables me to build and upload the code, check that it all works, and communicate using SCPI. All good, and works fine - except....
I expect that certain items will fail self test because they are not there. The display of course works. The RTC works too and keeps time. Many other things work, but the SD card fails self test.
Also the display shows ERROR and does not get past this. I did expect a full but non-functional display.
Is this expected behaviour? Will the SD card work when I get connected to the real power supplies?
I like to check stuff as I progress with the build, so what can I expect to be working just running off the native usb port?
Thanks.

prasimix, 2020-01-18:
SD card test is not related to power modules but to SD card itself. Check if SD card is functional on e.g. PC that accept SD cards. Also take into account that display's SD card socket may has no contact or that SD card is not inserted properly.

jfitter, 2020-01-18:
mmm I have fitted an SD card extender cable and brought the socket to the front panel. Maybe the cable is too long. I will check it out. Thanks.

jfitter, 2020-01-20:
Still somewhat confused.
I have two extender cables - short and long - 100mm and 480mm.
I have two LCDs from different manufacturers with different controllers.
All combinations of extender cables and SD card readers work just fine, even with the readers on the end of long USB cables.
All combinations of extender cables and display mounted SD card sockets fail, always.
SD cards inserted directly into display mounted SD card sockets work, always.
An SD card pried apart and carefully soldered to the end of an extender cable fails (to eliminate physical differences between the extender and a real SD card).

There is something about an extender cable that only the display card is incompatible with, regardless of the length of the cable.

Is this likely to be solved in software (timing etc.)?

Any hints would be appreciated.

I think I am missing something here. Is the SD card interface on the LCD display non standard? Is this a drive level issue? Do I need buffers? If so then why does the extender cable work connected to a reader connected to a PC on a 2m USB cable?

prasimix, 2020-01-20:
Communication with SD card on the LCD display is done via SPI. I don't know how USB cable came into picture.

jfitter, 2020-01-20:
SOLVED
The TFT_320QVT display has three 1K resistors in series with the signal lines to the SD card socket - LCD_CS, SCLK, and MOSI. I presume they are there to allow the display to be used with 5V signalling, say with a 5V Arduino. I replaced them with 0R shunts and it now works.

The clue came from the guys on the 3D printer forums who like to extend their SD cards to the front panel. Only the ones extended from the back of the LCD displays failed to work. As far as I can ascertain they have not yet solved their problem.

So, the resistors loaded up the signal lines enough so no amount of extension worked. The only arrangement that worked is if the SD card was actually in the socket. That's a bit of marginal engineering I would say - to save the cost of a buffer.

Maybe this will save someone a lot of time and head scratching.

closed 2020-01-20 by jfitter